adafruit/Adafruit_Protomatter #10: How to setup with Metro M0?

wma12, 2020-05-13:
Hi,

I have tried to use the RGB-Matrix-Panel library to work with multiple 32x16 matrices, unfortunately it seems like it is not possible to chain more than 2 of them. This is why I am looking for an alternative library for my Adafruit Metro M0. From the description this seems to be what I am looking for.

However I do not fully understand what I have to do to set it up for the Metro M0. I have connected it the same way I did with the RGB-Matrix-Panel library (using an Adafruit RGB Matrix Shield). I can succesfully compile and upload the protomatter example, but it won't display anything.

I probably need to somehow tell which Microcontroller I am using? I guess it has something to do with these lines:

#elif defined(_SAMD21_)
  uint8_t rgbPins[]  = {6, 7, 10, 11, 12, 13};
  uint8_t addrPins[] = {0, 1, 2, 3};
  uint8_t clockPin   = SDA;
  uint8_t latchPin   = 4;
  uint8_t oePin      = 5;

But where do I define it?

Or what else could be the problem?

PaintYourDragon, 2020-05-13:
You can delete all those ifdef/elif sections (those are all for testing on different Feather boards) and instead try these for Metro M0 + RGB Matrix shield:

uint8_t rgbPins[] = {2, 3, 4, 5, 6, 7};
uint8_t addrPins[] = {A0, A1, A2, A3};
uint8_t clockPin = 8;
uint8_t latchPin =10;
uint8_t oePin = 9;

wma12, 2020-05-14:
Thank you very much for the quick response. At first this did not solve the problem, or it caused a new one. I did the changes and it seemed to compile and upload without any problems, but immediately after uploading I see a Windows message "USB Device not detected". And from then on I didn't see the port for the Metro M0 anymore.

I needed to turn on verbose mode for uploading and reset the Metro M0 to get a USB connection again.

But there was one time where it actually worked, before it disrupted the connection again. I did some testing and realised that I need to turn down the bit depth from 4 to 3. Also I noticed that it was set up for a matrix with a height of 32px instead of the 16px mine has. This solved the problem with the interrupted connection.

When adding more matrices the problem occured again, but after turning down the bit depth even more to 2, I was able to create a scrolltext over a 192x16 matrix with double buffering turned on. The low bit depth is no problem for me.

I am really happy, that it finally worked. I was about to start over again and use a raspberry pi instead.

So my guess is, that the connection problems occrus when too much RAM is used and this somehow crashes the Metro M0? Or why would the size of the matrix and the bit depth have an effect on the USB connection?

Thank you very much :)
Maybe this also helps someone else who runs into similar problems.

PaintYourDragon, 2020-05-14:
In certain configurations the matrix will use a lot of RAM, yeah. USB went to heck because there’s no protected memory and potentially anything could get clobbered.
Quick/easy option is to look at the return value from matrix.begin(). It’ll return PROTOMATTER_ERR_MALLOC if things won’t fit.
Underlying issue is that the shield was designed for the Arduino UNO, but the newer library really wants to be on specific pins for optimal RAM usage…as it is, it’s using about 4X as much. Alternative would be wiring up something custom, or using the Feather M4 + Matrix FeatherWing combo (and, for similar reasons, would have to be the Feather M4 specifically, not the M0).

closed 2020-05-15 by ladyada

wma12, 2020-05-25:
I have one more question. I would like to connect the Metro M0 to another shield that needs the SPI Pins (D2 and D10-D13). Therefore I need to free up D10 and D2.
If I make some modifications to the RGB Matrix Shield, could I just use D1 as Latch Pin and D0 as the first RGB-Pin instead? 
Like so:

uint8_t rgbPins[] = {0, 3, 4, 5, 6, 7};
uint8_t addrPins[] = {A0, A1, A2, A3};
uint8_t clockPin = 8;
uint8_t latchPin =1;
uint8_t oePin = 9;

Or could this cause problems since D0 and D1 are the Pins for serial communication and I flash the Arduino over USB? Or could I use the free analog Pins A4 and A5?

PaintYourDragon, 2020-05-25:
If the shield uses the 6-pin ISCP header for SPI, then it’s probably OK still using 10 & 12 for the matrix. Depends how the shield’s designed.

On AVR-based boards like the Arduino Uno, SPI signals are routed both to pins 11-13 and that 6-pin header (the one at the end of the board OPPOSITE the USB plug, not the one that’s right next to it…though that one’s not populated on a lot of boards anyway). On ARM-based boards like the Metro or Arduino Zero, different signals are routed to each…the “true” hardware SPI is on the 6-pin header.

Early Arduino shields, or some not-well-thought-out ones, might only use 11-13 for SPI. Better ones will have a 6-pin female plug on the underside that interfaces with that header. BUT…it’s possible some of those might themselves route signals to both sets of pins. If a schematic is available, that might have answers, or could test with a multimeter with the shield not plugged into a board. If they do route to both, these could be fixed by cutting traces.