adafruit/Adafruit_Protomatter #17: STM32: Fix timer frequency

jepler, 2020-08-13:
We merged #14 though there is some doubt as to whether _PM_timerFreq is right for CircuitPython / STM32.  Investigate and ensure that the value is correct, possibly by making the macro expand to an appropriate function call or global variable.

@hierophect I can't assign this to you but it would be natural for you to look at this when you have time.

hierophect, 2020-08-13:
No problem, thanks for the ping. I've been distracted by ESP32-S2 stuff but I'll circle back to this as soon as I can, it's at the top of my bugfix list.

ladyada-eagleclaw, 2026-10-02:
A minimal fix is ready in #95: query the actual clock source of the selected timer instead of assuming 42 MHz. The STM32F405 timers run at 84 MHz on APB1 or 168 MHz on APB2 when PSC=0.

The complete Feather STM32F405 Express CircuitPython firmware builds, and all six Arduino examples still compile on MatrixPortal S3. This is compile-validated; no STM32 board was available for a physical panel test. The PR closes this issue when merged.

closed 2026-10-02 by ladyada