adafruit/Adafruit_Protomatter #27: Updating Pixeldust example to support things like Feather nRF52840 Sense

justmobilize, 2020-10-27:
Hello!

I have created an updated example for Pixeldust, to support the Feather Sense and a few wings that have an accelerometer. I find it super helpful for making sure I got a board soldered well (obviously doesn't verify all the pins).

I was wondering if you would want a PR?

And if so, if you would want the main example updated, or ones for a specific board / feather

ladyada, 2020-10-27:
a single example just for feather sense would be best :)

justmobilize, 2020-10-28:
Will do!

ladyada-eagleclaw, 2026-10-02:
The current PixelDust example still targets MatrixPortal and LIS3DH. It does not implement the separate Feather Sense example requested here.

A Feather Sense sketch remains the intended scope, using that board's IMU and the RGB Matrix FeatherWing pin map. The current S3 bench tests cannot validate the Sense accelerometer/orientation, so this remains open as an example feature request.

Current example: https://github.com/adafruit/Adafruit_Protomatter/blob/0f51ca2347479228d8cb96fd3f410b032fe6dbad/examples/pixeldust/pixeldust.ino