adafruit/Adafruit_Protomatter #40: Issue compiling animated_gif example (with solution)

Simonalexanderadams, 2021-06-24:
I encountered an issue compiling the animaged_gif example. It was solved, as noted on[ this forum thread](https://forums.adafruit.com/viewtopic.php?f=19&t=171857&p=878634), by changing line 92 to
`void *GIFOpenFile(const char *filename, int32_t *pSize) {`

I was able to find the solution this after some digging, but I would suggest updating the example file so it doesn't continue to occur for others in the future.

ladyada, 2021-06-24:
thanks, please submit a PR - we can merge it!

Simonalexanderadams, 2021-06-24:
Can do. That said, I may have spoken too soon. While the _compiling_ was a success, I wasn't able to get it to work correctly with my hardware (Matrix Portal M4 and an Adafruit 64x32 LED matrix.) Currently I've gotten the python based [sprite sheet example ](https://learn.adafruit.com/pixel-art-matrix-display/overview)working, and have been able to successfully run the simple Protomatter example, but haven't had successful gif playback. My issue seems like more of a forum question, though since I have other examples running I probably won't be bringing it up in the interest of time.

Let me know if you'd still like me to create the PR, though I'd still suggest doing a hardware test on your end before merging since I can't verify that it actually works. (The solution has appeared to work for others on the forum at least.)

ladyada-eagleclaw, 2026-10-02:
The requested callback signature is already present on current master:

```cpp
void *GIFOpenFile(const char *filename, int32_t *pSize)
```

The `animated_gif` example compiles with Arduino-ESP32 3.3.11 on MatrixPortal S3. Closing the reported callback compilation error as resolved. This does not claim to resolve the separate playback trouble mentioned in the comments; long-GIF truncation is tracked in #90. A separate TinyUSB dependency issue affects our Arduino-ESP32 4.0.0-alpha1 build.

closed 2026-10-02 by ladyada-eagleclaw