adafruit/Adafruit_Protomatter #42: matrix.show() not working in interrupt routine

kanadagermane, 2021-11-09:
Hey,

I am using this library with an Adafruit MatrixPortal M4 and PlatformIO. 
Since I do blocking HTTP requests with WiFiNINA (adafruit fork), I thought I can update the matrix in an timer interrupt to be able to do a scollable text. 

I am using the [SAMD_TimerInterrupt](https://github.com/khoih-prog/SAMD_TimerInterrupt) library for the interrupts. Unfortunately, the program execution crashes if I call matrix.show() in the TimerHandler. The setCursor() and print() methods do not lead to a crash.

`void TimerHandler0()
{
  static uint32_t curMillis = 0;

  curMillis = millis();
  if (curMillis > TIMER0_INTERVAL_MS)
  {
      matrix.fillScreen(0);
      matrix.setCursor(textX, textY);
      matrix.print(str);
      if ((--textX) < textMin)
        textX = matrix.width();
      matrix.show();
  }
  preMillisTimer0 = curMillis;
}`

Please let me know if you need anything additional from me. 
Maybe there is even a better way of doing HTTP requests during an scrolltext, but I was not able to find a way of doing async http requests with the MatrixPortal.

Thanks in advance and best regards,
Daniel

aiweier1231, 2022-01-20:
Hi kanadagermane,

You can integrate FreeRTOS or any real time OS to ur project ot achieve this. Making HTTP request and LED Matrix display two separate tasks.

kanadagermane, 2022-01-21:
Thank you so much for this reply, you made my day!
For sure that was exactly what I was missing.

closed 2022-01-21 by kanadagermane