adafruit/Adafruit_Protomatter #49: hi now some P4 panel comes with 80x40 led matrix with 1/20 scan ...and unfortunately i have one ..how to run them ..i have hooked and found only 32 out of 40 rows are working

ravigyanvihar, 2022-07-01:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

board707, 2022-07-16:
Hi @ravigyanvihar 
Few month ago I have modified the old variant of the library - Adafruit RGB-matrix-Panel - to add support for 80x40 RGB matrices. But the code was only tested on Arduino Mega.  My fork of  RGB-matrix-Panel  you can see at [https://github.com/board707/RGB-matrix-Panel/tree/matrix80x40](https://github.com/board707/RGB-matrix-Panel/tree/matrix80x40) Make sure to select the "matrix80x40" branch before downloading. For usage see the example Mega_matrix_80x40.ino"

If you interesting to use your matrix with Arduino STM32 boards - I have a custom library for RGB and Monochrome DMD panels which supported many kinds of matrices, including RGB 80x40 20 scan. See the link [DMD_STM32 arduino library](https://github.com/board707/DMD_STM32)

priya2212, 2023-01-10:
@board707 , Thank you for providing the library for matrix80x40. This really eases our work. I am using 80x40 P4 led. I am not sure about scan rate as I got it from local vendor and they don't have any reference for the same. I am assuming the scan rate is 1:10.
However, when I am running the code, and drawing the pixel at x=0, y=0, getting the pixel output at x=0,y=0 and x=0, y=10. Same is occurring with Text. 
Would be great if you can give me your valuable inputs.
[image]
[image]

board707, 2023-01-12:
HI
You wrote:

> I am assuming the scan rate is 1:10.

Sorry, but my code will work with 80x40 scan20 panels only.
Your matrix is a tricky case. A standard 80x40 matrix has 1:20 scans, and you have 1.10. This means that the LEDs in your matrix are connected in a non-standard way.
I can try to run your matrix, but only using STM32 or RP2040 boards, using my library [DMD_STM32](https://github.com/board707/DMD_STM32)

ravigyanvihar, 2023-02-26:
i changed number of lines and solved it

closed 2023-02-26 by ravigyanvihar