adafruit/Adafruit_Protomatter #66: RGBMatrix/Protomatter with Metro M7 iMXRT1011

DJDevon3, 2023-09-07:
Purely as an experiment I attempted to use a Metro M7 with the [RGB Matrix Shield](https://www.adafruit.com/product/2601) in Circuit Python and it comes back with `ImportError: no module named 'rgbmatrix'`.  

Was advised in discord dev chat to make a feature request in Protomatter... even though I'm attempting to use it in Circuit Python? Attempted to add MatrixPortal and PortalBase libraries as a work around and it ran out RAM.  

[image]

Maybe the M7 won't even run matrix panels? It's above my head and is a better idea for the devs to discuss if this is or is not a feasible method vs sticking with a Matrix Portal S3.  

I have 6 panels running on the Matrix Portal S3... was just trying to see if the CPU of the iMX would make a positive difference.

makermelissa, 2023-09-07:
MatrixPortal and PortalBase rely on the RGBMatrix module. I'm kind of surprised it didn't fail on the import unless it just has very little RAM.

DJDevon3, 2023-09-08:
My thought process was if there's support for the M4 with that shield then it might work on the M7 iMX.  It didn't even get past the import for RGBmatrix. 

It did fail on import of the matrixportal and portalbase with a ram allocation error.  I know those are rather large libraries which is why I try to run only the rgbmatrix library. 

On the 6 panels I have I'm only running the rgbmatrix library.  It cuts out a lot of the overhead. [128x96 MatrixPortal S3 demo](https://github.com/DJDevon3/My_Circuit_Python_Projects/tree/main/Boards/espressif/Adafruit%20MatrixPortal%20S3/128x96%20RGB%20Matrix).

makermelissa, 2023-09-08:
True, but you don't need to import the top level (see https://learn.adafruit.com/creating-projects-with-the-circuitpython-matrixportal-library/choosing-your-modules). For the message board, I only used the matrix layer to handle the initialization.

DJDevon3, 2023-09-09:
I've never seen that layer map before.  Was unaware you could cherry pick layers like that, that helps! Thank you. I'll try `from adafruit_matrixportal.graphics import Graphics` which should give me access to rgbmatrix and displayio.  Have my doubts much if any ram will be left over for buffer.

Currently have all the panels disassembled and working on 3D printing brackets. Will have to revisit this after it's structurally sound as 1 large panel.  Wish I could use your bracket design but I'm using the 5mm pitch panels which are physically different and in a different configuration so have to design my own.

ladyada-eagleclaw, 2026-10-02:
Metro M7 RT1011 still lacks the native CircuitPython rgbmatrix implementation. The existing Protomatter i.MX backend is guarded for RT1062 and implements Arduino/Teensy; its CircuitPython section is empty. Installing Python MatrixPortal libraries cannot supply that missing firmware module.

This remains an RT1011/CircuitPython port request, requiring backend integration and hardware validation. The recent ESP32-S3/IDF6 work does not add this port.

Backend: https://github.com/adafruit/Adafruit_Protomatter/blob/0f51ca2347479228d8cb96fd3f410b032fe6dbad/src/arch/teensy4.h