adafruit/Adafruit_Protomatter #69: Possible `_PM_esp32commonTimerInit` bug?

J-MR-T, 2023-10-18:
I just tried to half-assedly port the S3 code I need to arduino-esp 3.0.0-alpha2, which uses a refactored timer API (https://github.com/espressif/arduino-esp32/pull/7904 and https://github.com/espressif/arduino-esp32/pull/7906), to get it to somewhat work, I mostly only had to substitute a few calls and numbers, but one change stood out to me, and I think it might be a bug in the current version too, so I'd like to bring it to someones attention ^^'.

I'm talking about [`_PM_esp32commonTimerInit`](https://github.com/adafruit/Adafruit_Protomatter/blob/429d625ba7a646d78f74402500eaad1e7b106aad/src/arch/esp32-common.h#L89-L95), specifically the [last line](https://github.com/adafruit/Adafruit_Protomatter/blob/429d625ba7a646d78f74402500eaad1e7b106aad/src/arch/esp32-common.h#L94C1-L94C1). Shouldn't this be core->timer? Because if the `if` statement above gets executed, `core->timer` is overwritten, but not the local `timer` variable, which still contains the old `core->timer` from before the `timerBegin` call.

When I changed the call in line 94 to use `core->timer`, my matrix started to light up, whereas something crashed along the way if I compile without that change. All on arduino-esp 3.0.0-alpha2.

I'm happy to provide more information if necessary.

Cheers :)

ladyada, 2023-11-14:
hiya yes we have to update this library to ESP 3 - thanks for the hint, we'll take a look!

ladyada, 2023-11-14:
also if you have a PR please submit it in case our port has bugs

ladyada-eagleclaw, 2026-10-02:
The reported stale timer pointer is fixed on current master: both Arduino timer API paths attach the interrupt to `core->timer`, including when initialization allocates a new timer.

Verified on MatrixPortal S3 with a standard 64x64 A-E panel using Arduino-ESP32 3.3.11 / ESP-IDF 5.5.5 and Arduino-ESP32 4.0.0-alpha1 / ESP-IDF 6.0.1. Both builds initialized correctly and completed 200 stop/resume cycles without a reset or visible corruption, using the code merged in #93.

Closing this specific timer-pointer report as resolved. Thanks for identifying it!

closed 2026-10-02 by ladyada-eagleclaw