adafruit/Adafruit_Protomatter #76: Annoying interference on 2xMatrix LED + Matrix Portal S3 using AIO

damomad12, 2024-04-26:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

dhalbert, 2024-04-26:
Could you please edit your post and replace the screenshot with text? For instance, we'd like to try your code, and it's only available as an image.

damomad12, 2024-04-26:
Hi. Thank you for your answer.
I tried to paste it, but it didn't work. I'll try again.
Succeeded to paste the text!
Here it is:

Annoying interference on 2xMatrix LED + Matrix Portal S3 using AIO
 
Hi.
I’m experiencing an annoying distortion on the Matrix LED displays. It’s like random horizontal lines (or points) appearing & disappearing. Always at the right side of any LEDs being on. It’s like a copy of the LEDs appearing at their right side.
 
The HW setup is a Matrix Portal S3 and 2 identical LED Matrix Displays 64x64. The MPS3 is directly connected to one of the displays which is chained to the other display.
 
- Arduino board:  Adafruit Matrix Portal S3
 
- Arduino IDE version (found in Arduino -> About Arduino menu):  2.3.2
 
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):
 
I have 3 different devices:
-              Device 1: MPS3 with 2 chained identical LED Matrix Display 64x64
-              Device 2: MPS3 with 2 chained identical (smaller) LED Matrix Displays 64x64
-              Device 3: MPS3 with 1 LED Matrix Display 64x32
 
The HW I’m using is all from Adafruit.
 
Although I noticed (on both devices 1 and 2) just a bit of this distortion about every 15 sec, it was not of importance. But when I configured & connected the MPS3 to the Internet and to AIO, the effect became very visible and quite annoying. This was on both MPS3 +2 chained displays, devices 1 and 2.
On device 3, the MPS3 single display, never any distortion.
 
I’ve been in touch with Adafruit’s support and we´ve tried many things to try to find out what was happening: changed the USB-C cable to support high intensity, although the device is consuming about 0,5A, change ribbon cables, configuration, etc.
 
Always using the same code, if I disconnect the second display, distortion disappears
If I use both 64x64 panels, but with a 64x32 SW setup. No interference!!! using: Adafruit_Protomatter matrix(64, 4, 1, rgbPins, 4, addrPins, clockPin, latchPin, oePin, false); If I use both 64x64 panels, but with 64x64 SW setup. Interference appears again using: Adafruit_Protomatter matrix(128, 4, 1, rgbPins, 5, addrPins, clockPin, latchPin, oePin, false);
 
Adafruit support conclusion is that the HW is ok:
“All hardware looks good. This seems like an issue with the Protomatter library since the problem only occurs when a unified address space is used between two panels.”
 
They have advised me to put this issue on this GitHub channel.
 
The effect can be seen on this video: https://youtu.be/DGt_X1TwFok
 
Thank you in advance.
 
Dany
 
The code I’m using is:
 
```
//sin rtc
//quitando más cosas innecesarias
/************************** INTERNET Configuration ***********************************/
 
// Internet AIO
// edit the config.h tab and enter your Adafruit IO credentials
// and any additional configuration needed for WiFi, cellular,
// or ethernet clients.
#include "config.h"
/************************** End INTERNET Configuration ***********************************/
 
#include <Adafruit_Protomatter.h>
#include <Adafruit_GFX.h>       // Core graphics library
 
// RGB565 Colour codes
#define ST7735_LIME    0x07FF
#define ST7735_RED     0xF800
 
// CONFIGURA AUTOMÁTICAMENTE ENTRE EL TIPO DE PLACA M4 Y S3
  uint8_t rgbPins[] = {42, 41, 40, 38, 39, 37}; // *****OK**** CAMBIOPANEL COLORES PARA 64X64 NÚMEROS ROJO, H/M AMARAILLO, FÓRMULA AZUL
  uint8_t addrPins[]  = {45, 36, 48, 35, 21};
  uint8_t clockPin =2;
[... 284 lines cut ...]
                   SPIWIFI_ACK, ESP32_RESETN, ESP32_GPIO0, &SPIWIFI);
#else
AdafruitIO_WiFi io(IO_USERNAME, IO_KEY, WIFI_SSID, WIFI_PASS);
#endif
/******************************* FONA **************************************/
 
// the AdafruitIO_FONA client will work with the following boards:
//   - Feather 32u4 FONA -> https://www.adafruit.com/product/3027
 
// uncomment the following two lines for 32u4 FONA,
// and comment out the AdafruitIO_WiFi client in the WIFI section
// #include "AdafruitIO_FONA.h"
// AdafruitIO_FONA io(IO_USERNAME, IO_KEY);
 
/**************************** ETHERNET ************************************/
 
// the AdafruitIO_Ethernet client will work with the following boards:
//   - Ethernet FeatherWing -> https://www.adafruit.com/products/3201
 
// uncomment the following two lines for ethernet,
// and comment out the AdafruitIO_WiFi client in the WIFI section
// #include "AdafruitIO_Ethernet.h"
// AdafruitIO_Ethernet io(IO_USERNAME, IO_KEY);
 
```

damomad12, 2024-04-26:
Oops, I accidently closed the issue. Sorry, I've opened it again

dhalbert, 2024-04-26:
Forums thread is here: https://forums.adafruit.com/viewtopic.php?t=210086

dhalbert, 2024-04-26:
Thank you for all your testing. You mentioned in the forum posts that if wifi is not in use, you don't see the interference. My guess is that wifi activity is interrupting Protomatter's dumping data to the panels. Protomatter uses up a lot of CPU time at interrupt level driving the panels, which have to be driven in real time with precise timing.  The more panels, the more time it takes. The wifi activity may be running at a higher priority, or perhaps at some point the wifi activity can no longer be deferred and takes priority. This is just speculation. (I am not necessarily the person who would work on this.)

damomad12, 2024-04-26:
Hi. Thanks for your reply.

As I told at the starting post: "Although I noticed (on both devices 1 and 2) just a bit of this distortion about every 15 sec (more or less), it was not of importance."
So there was an ocasional interference from time to time, but not that important.

Ok I understand your guess. Is there anyway I can lower the priority or the wifi activity or Protomatter 's dumping data, so I can find any improvement?

On the other hand, if I tell Protomatter that it's displays width is 64 instead of 128, no interference appears, although the same info is seen on both screens. I suppose that's due to the fact that it takes double time to send info to the 128 screen instead of a 2x64. Could this be right?

If you are not necessarily the person to work on this, will there be anyone else who may help me?

Thank you for your support.

Dany.

th-kwon, 2024-06-18:
Hello. We're experencing simillar issue from S3 with large matrix.
Using sample code from protomatter(doublebuffer_scrolltext) changing some parameter reproduces this issue.

Also this interference is not occuring from MatrixPortal M4 with same code.

Scrolltext from S3 - tearing occurs(0:40)
https://youtu.be/IvfrAela4C0
[image]

Scrolltext from M4 - no tearing
https://youtu.be/ZV2n-bKIZ2A

What we're going to do - displaying large gif images and some text message. 
https://youtu.be/_q8w-LxM-3Q
We're looking for using S3 than M4 for larger ram and flash size, but this issue needs to be resolved before using it.

The code below is sample code what we used from video
(width 192, height 64, bitdepth 4, text size 2)

```c
/* ----------------------------------------------------------------------
Double-buffering (smooth animation) Protomatter library example.
PLEASE SEE THE "simple" EXAMPLE FOR AN INTRODUCTORY SKETCH.
Comments here pare down many of the basics and focus on the new concepts.

This example is written for a 64x32 matrix but can be adapted to others.
------------------------------------------------------------------------- */

#include <Adafruit_Protomatter.h>
#include <Fonts/FreeSansBold18pt7b.h> // Large friendly font

/* ----------------------------------------------------------------------
The RGB matrix must be wired to VERY SPECIFIC pins, different for each
microcontroller board. This first section sets that up for a number of
supported boards.
------------------------------------------------------------------------- */

#if defined(_VARIANT_MATRIXPORTAL_M4_) // MatrixPortal M4
  uint8_t rgbPins[]  = {7, 8, 9, 10, 11, 12};
  uint8_t addrPins[] = {17, 18, 19, 20, 21};
  uint8_t clockPin   = 14;
  uint8_t latchPin   = 15;
  uint8_t oePin      = 16;
#elif defined(ARDUINO_ADAFRUIT_MATRIXPORTAL_ESP32S3) // MatrixPortal ESP32-S3
  uint8_t rgbPins[]  = {42, 41, 40, 38, 39, 37};
  uint8_t addrPins[] = {45, 36, 48, 35, 21};
  uint8_t clockPin   = 2;
  uint8_t latchPin   = 47;
  uint8_t oePin      = 14;
#elif defined(_VARIANT_FEATHER_M4_) // Feather M4 + RGB Matrix FeatherWing
  uint8_t rgbPins[]  = {6, 5, 9, 11, 10, 12};
  uint8_t addrPins[] = {A5, A4, A3, A2};
  uint8_t clockPin   = 13;
  uint8_t latchPin   = 0;
  uint8_t oePin      = 1;
#elif defined(__SAMD51__) // M4 Metro Variants (Express, AirLift)
  uint8_t rgbPins[]  = {6, 5, 9, 11, 10, 12};
  uint8_t addrPins[] = {A5, A4, A3, A2};
  uint8_t clockPin   = 13;
  uint8_t latchPin   = 0;
  uint8_t oePin      = 1;
#elif defined(_SAMD21_) // Feather M0 variants
  uint8_t rgbPins[]  = {6, 7, 10, 11, 12, 13};
  uint8_t addrPins[] = {0, 1, 2, 3};
  uint8_t clockPin   = SDA;
  uint8_t latchPin   = 4;
  uint8_t oePin      = 5;
#elif defined(NRF52_SERIES) // Special nRF52840 FeatherWing pinout
  uint8_t rgbPins[]  = {6, A5, A1, A0, A4, 11};
  uint8_t addrPins[] = {10, 5, 13, 9};
[... 139 lines cut ...]
  // Update text position for next frame. If text goes off the
  // left edge, reset its position to be off the right edge.
  if((--textX) < textMin) textX = matrix.width();

  // Draw the three bouncy balls on top of the text...
  for(byte i=0; i<3; i++) {
    // Draw 'ball'
    matrix.fillCircle(ball[i][0], ball[i][1], 5, ballcolor[i]);
    // Update ball's X,Y position for next frame
    ball[i][0] += ball[i][2];
    ball[i][1] += ball[i][3];
    // Bounce off edges
    if((ball[i][0] == 0) || (ball[i][0] == (matrix.width() - 1)))
      ball[i][2] *= -1;
    if((ball[i][1] == 0) || (ball[i][1] == (matrix.height() - 1)))
      ball[i][3] *= -1;
  }

  // AFTER DRAWING, A show() CALL IS REQUIRED TO UPDATE THE MATRIX!

  matrix.show();

  delay(20); // 20 milliseconds = ~50 frames/second
}
```

dandindarov, 2024-07-10:
Hi, I'm experiencing the same issue using CircuitPython. Same exact sparkles as the videos above.

I am also running a bigger than usual matrix display at 128x64 (non-adafruit)
I also had no issues when using a MatrixPortal M4 - the issues only appear on a MatrixPortal S3. Maybe the M4 draws less power?
Running the exact same code on both boards.

My issue also stops if I use only half of the display at 64x64, or use less bit depth.
I had a little venture and attempted to underclock the CPU to try and draw less power. This actually seems to fix my problem but then introduces an overall flicker/interferance instead of just a few leds sparkling.

I was told on discord after discussing this that my power might not be enough. I do personally think it is indeed power related. So I will attempt to power the matrix separately with plenty of supply and report back.

damomad12, 2024-07-10:
-- > Hi, I finally solved the issue, take a look at the Adafruit Forum (https://forums.adafruit.com/viewtopic.php?t=210086), were I continued receiving support.
I have not tested it with MP M4, although I read somewhere that other people had less issues with this board. Maybe S3 is less powerful than M4? 

> Hi, I'm experiencing the same issue using CircuitPython. Same exact sparkles as the videos above.
>
> I am also running a bigger than usual matrix display at 128x64 (non-adafruit) I also had no issues [...]
--> I solved it almost with bit depth of 2, and completely with depth bit of 1
> I was told on discord after discussing this that my power might not be enough, I'm using a 5V 5A 27W. I do personally think it is indeed power related. So I will attempt to power the matrix separately with plenty of supply and report back.
--> I was also suggested that it could be a power issue, but I had more than enough power. Just check your power consupmtion in Amp and if it's significantly less than your power adaptor may deliver, this shouldn't be the issue. Your 5A could be enough depending on how many leds you turn on....
Try first with bit depth of 1!

dandindarov, 2024-07-11:
Unfortunately lowering bitdepth (which does indeed work) is not a viable solution for me as I need different levels of brightness across the screen.

th-kwon, 2024-07-12:
I solved my problem by using another library [ESP32-HUB75-MatrixPanel-DMA](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA). Flickerings are gone after changing it.

I'm assuming this problem is related to memory consumption because I can reproduce this issue regardless of number of connected panels, only affects with width, height and bit depth option. (using small panel but setup with large width like 128x64 makes same flickering)

dandindarov, 2024-07-13:
Thanks for this! My project is unfortunately all written in CircuitPython using this library (Protomatter/rgbmatrix)
It would be interesting to do some tests in Arduino with that library.

But I'd like to keep my project in CircuitPython.

After my power tests it doesn't look like it is power related unfortunately.

Great you linked that, maybe the expert devs could look into that library and see what we're missing/is so different from the one we have here in Protomatter?

dandindarov, 2024-10-17:
Just checking if any devs had time to have a look at this? cheers

ladyada-eagleclaw, 2026-10-02:
Retested on MatrixPortal S3 with current master (0f51ca2, including #93), four-bit depth, double buffering, one physical standard 64x64 panel and a 128x64 logical canvas. The wider logical canvas follows the single-panel reproduction described above. Sparse pixels kept the USB-powered test load low.

- Arduino-ESP32 4.0.0-alpha1 / IDF 6: 60 HTTP downloads, 6,960,240 bytes received, no reset.
- Arduino-ESP32 3.3.11 / IDF 5.5.5: 40 complete downloads, then a download timeout; this run did not complete the planned 60.
- Default Wi-Fi power saving was retained. Sampled camera frames did not show stray pixels.

Keeping this open: this was not the original two-panel Adafruit IO workload, and sampled camera frames cannot rule out every brief sparkle. A retest of the original failing setup with current master would help establish whether the transfer-completion fixes resolve it. The IDF 5.5 network timeout is also not being counted as a fully passing stress run.