adafruit/Adafruit_Protomatter #78: Not compatible with esp32 version 3.0.0

chockenberry, 2024-06-01:
I recently updated to version 3.0.0 of Espressif's 'esp32' package (after being prompted to by the IDE).

When building Protomatter with this new version, I see the following errors:

```
In file included from /Users/CHOCK/Documents/Arduino/libraries/Adafruit_Protomatter/src/arch/arch.h:190,
                 from /Users/CHOCK/Documents/Arduino/libraries/Adafruit_Protomatter/src/core.c:33:
/Users/CHOCK/Documents/Arduino/libraries/Adafruit_Protomatter/src/arch/esp32-common.h: In function '_PM_esp32commonTimerInit':
/Users/CHOCK/Documents/Arduino/libraries/Adafruit_Protomatter/src/arch/esp32-common.h:92:19: error: too many arguments to function 'timerBegin'
   92 |     core->timer = timerBegin(_PM_timerNum, 2, true); // 1:2 prescale, count up
      |                   ^~~~~~~~~~
In file included from /Users/CHOCK/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/esp32-hal.h:84,
                 from /Users/CHOCK/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/Arduino.h:36,
                 from /Users/CHOCK/Documents/Arduino/libraries/Adafruit_Protomatter/src/arch/arch.h:166:
/Users/CHOCK/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
/Users/CHOCK/Documents/Arduino/libraries/Adafruit_Protomatter/src/arch/esp32-common.h:94:3: error: too many arguments to function 'timerAttachInterrupt'
   94 |   timerAttachInterrupt(timer, &_PM_esp32timerCallback, true);
      |   ^~~~~~~~~~~~~~~~~~~~
/Users/CHOCK/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~

exit status 1
```

The workaround is to go into the Boards Manager, find 'esp32', and downgrade to version 2.0.17.

lboue, 2024-07-26:
Same issue for me. Could you fix this?

biomurph, 2024-07-27:
Hi,
An issue came up here
https://github.com/WorldFamousElectronics/PulseSensorPlayground/issues/195
Similar to the one you are posting about. 

Does the other thread solve your problem?

lboue, 2024-07-27:
> Hi, An issue came up here [WorldFamousElectronics/PulseSensorPlayground#195](https://github.com/WorldFamousElectronics/PulseSensorPlayground/issues/195) Similar to the one you are posting about.
> 
> Does the other thread solve your problem?

No but I found a fix. You can download [Adafruit_Protomatter-master.zip](https://github.com/adafruit/Adafruit_Protomatter/archive/refs/heads/master.zip) And install it manually in the Arduino IDE.

biomurph, 2024-07-29:
@chockenberry @lboue 
This comes down to a protomatter issue, so I am closing it here.

biomurph, 2024-08-08:
@lboue 
I am getting a runtime error: 
`E (127) gptimer: gptimer_start(348): timer is not enabled yet`
 I have more notes to edit into this comment, but it seems the timer is not turned on?
I am targeting ESP32-S2 Dev Generic.

codeforlibre, 2024-08-27:
> > Hi, An issue came up here [WorldFamousElectronics/PulseSensorPlayground#195](https://github.com/WorldFamousElectronics/PulseSensorPlayground/issues/195) Similar to the one you are posting about.
> > Does the other thread solve your problem?
> 
> @biomurph No but I found a fix. You can download [Adafruit_Protomatter-master.zip](https://github.com/adafruit/Adafruit_Protomatter/archive/refs/heads/master.zip) and install it manually in the Arduino IDE. The patch is already in the repo but has not been released in a new version.

@biomurph Hello, This fixed my "error: too many arguments to function 'timerBegin' 92 | core->timer = timerBegin(_PM_timerNum, 2, true); // 1:2 prescale, count up".
Arduino IDE 2.3.2
Board: Adafruit Metro ESP32-S2
@lboue Thank you for reporting the issue on Github. https://github.com/adafruit/Adafruit_Protomatter/issues/79

biomurph, 2024-08-27:
@codeforlibre 
🤘🏻

CyanBlob, 2024-09-06:
> @lboue I am getting a runtime error: `E (127) gptimer: gptimer_start(348): timer is not enabled yet` It seems the timer is not turned on? Or not started? After that serial message, it sends the following:

Are you still seeing this error? I'm running into this as well

closed 2024-09-23 by tannewt