adafruit/Adafruit_Protomatter #79: Could you please publish the 1.6.2 library version?

lboue, 2024-07-27:
Hello,

Could you please publish the 1.6.2 library version? Indeed the last commit ca4f876 was 3 months ago.
https://github.com/adafruit/Adafruit_Protomatter/blob/ca4f8764bedea915d4b9d85df1b57b6860bd7103/library.properties#L2

This version fixes this issue: [Not compatible with esp32 version 3.0.0 #78](https://github.com/adafruit/Adafruit_Protomatter/issues/78)

- Arduino board:  ADAFRUIT_MATRIXPORTAL_ESP32S3

- Arduino IDE version (found in Arduino -> About Arduino menu):  2.3.2

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): Build pixeldust.ino

lboue, 2024-08-16:
Hello,

Can I have a response?

Regards

tannewt, 2024-09-23:
I've published 1.7.0

closed 2024-09-23 by tannewt