adafruit/Adafruit_Protomatter #87: Errors with arduino-esp32 3.3.0 but it works fine with arduino-esp32 3.2.0

lboue, 2025-09-01:
Hi,

I am facing errors arduino-esp32 3.3.0 but it works fine with arduino-esp32 3.2.0.
Do you know how to fix errors?
```
FQBN: esp32:esp32:adafruit_matrixportal_esp32s3
Using board 'adafruit_matrixportal_esp32s3' from platform in folder: C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0
Using core 'esp32' from platform in folder: C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0
...
Compiling libraries...
Compiling library "Wire"
"C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp-x32\\2411/bin/xtensa-esp32s3-elf-g++" -MMD -c "@C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/flags/cpp_flags" -Wall -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ADAFRUIT_MATRIXPORTAL_ESP32S3 -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ADAFRUIT_MATRIXPORTAL_ESP32S3\"" "-DARDUINO_VARIANT=\"adafruit_matrixportal_esp32s3\"" -DARDUINO_PARTITION_tinyuf2-partitions-8MB "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:adafruit_matrixportal_esp32s3:UploadSpeed=921600,USBMode=default,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=cdc,CPUFreq=240,FlashMode=qio,FlashSize=8M,PartitionScheme=tinyuf2,DebugLevel=none,PSRAM=enabled,LoopCore=1,EventsCore=1,EraseFlash=none,ZigbeeMode=default\"" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=0 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/flags/defines" "-IC:\\Users\\ludov\\Downloads\\pixeldust" -iprefix "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/include/" "@C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/flags/includes" "-IC:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/qio_qspi/include" "-IC:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\cores\\esp32" "-IC:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\variants\\adafruit_matrixportal_esp32s3" "-IC:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\libraries\\Wire\\src" "-Ic:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_LIS3DH" "-IC:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\libraries\\SPI\\src" "-Ic:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_BusIO" "-Ic:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_Unified_Sensor" "-Ic:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_PixelDust" "-Ic:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_Protomatter\\src" "-Ic:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library" "@C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\303BA53C140C7AA0D0F4DF74DDB0D784/build_opt.h" "@C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\303BA53C140C7AA0D0F4DF74DDB0D784/file_opts" "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\libraries\\Wire\\src\\Wire.cpp" -o "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\303BA53C140C7AA0D0F4DF74DDB0D784\\libraries\\Wire\\Wire.cpp.o"
<command-line>: warning: ISO C++11 requires whitespace after the macro name
Compiling library "Adafruit LIS3DH"
"C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp-x32\\2411/bin/xtensa-esp32s3-elf-g++" -MMD -c "@C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/flags/cpp_flags" -Wall -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ADAFRUIT_MATRIXPORTAL_ESP32S3 -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ADAFRUIT_MATRIXPORTAL_ESP32S3\"" "-DARDUINO_VARIANT=\"adafruit_matrixportal_esp32s3\"" -DARDUINO_PARTITION_tinyuf2-partitions-8MB "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:adafruit_matrixportal_esp32s3:UploadSpeed=921600,USBMode=default,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=cdc,CPUFreq=240,FlashMode=qio,FlashSize=8M,PartitionScheme=tinyuf2,DebugLevel=none,PSRAM=enabled,LoopCore=1,EventsCore=1,EraseFlash=none,ZigbeeMode=default\"" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=0 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/flags/defines" "-IC:\\Users\\ludov\\Downloads\\pixeldust" -iprefix "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/include/" "@C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/flags/includes" "-IC:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/qio_qspi/include" "-IC:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\cores\\esp32" "-IC:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\variants\\adafruit_matrixportal_esp32s3" "-IC:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\libraries\\Wire\\src" "-Ic:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_LIS3DH" "-IC:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\libraries\\SPI\\src" "-Ic:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_BusIO" "-Ic:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_Unified_Sensor" "-Ic:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_PixelDust" "-Ic:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_Protomatter\\src" "-Ic:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library" "@C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\303BA53C140C7AA0D0F4DF74DDB0D784/build_opt.h" "@C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\303BA53C140C7AA0D0F4DF74DDB0D784/file_opts" "c:\\Users\\ludov\\Documents\\Arduino\\libraries\\Adafruit_LIS3DH\\Adafruit_LIS3DH.cpp" -o "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\303BA53C140C7AA0D0F4DF74DDB0D784\\libraries\\Adafruit_LIS3DH\\Adafruit_LIS3DH.cpp.o"
<command-line>: warning: ISO C++11 requires whitespace after the macro name
Compiling library "SPI"
"C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp-x32\\2411/bin/xtensa-esp32s3-elf-g++" -MMD -c "@C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/flags/cpp_flags" -Wall -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ADAFRUIT_MATRIXPORTAL_ESP32S3 -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ADAFRUIT_MATRIXPORTAL_ESP32S3\"" "-DARDUINO_VARIANT=\"adafruit_matrixportal_esp32s3\"" -DARDUINO_PARTITION_tinyuf2-partitions-8MB "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:adafruit_matrixportal_esp32s3:UploadSpeed=921600,USBMode=default,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=cdc,CPUFreq=240,FlashMode=qio,FlashSize=8M,PartitionScheme=tinyuf2,DebugLevel=none,PSRAM=enabled,LoopCore=1,EventsCore=1,EraseFlash=none,ZigbeeMode=default\"" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=0 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/flags/defines" "-IC:\\Users\\ludov\\Downloads\\pixeldust" -iprefix "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/include/" "@C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32s3/flags/includes" "-IC:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\ [... cut ...]
Using library Adafruit PixelDust at version 1.1.3 in folder: C:\Users\ludov\Documents\Arduino\libraries\Adafruit_PixelDust 
Using library Adafruit Protomatter at version 1.7.0 in folder: C:\Users\ludov\Documents\Arduino\libraries\Adafruit_Protomatter 
Using library Adafruit GFX Library at version 1.12.1 in folder: C:\Users\ludov\Documents\Arduino\libraries\Adafruit_GFX_Library 
exit status 1

Compilation error: exit status 1
```

[adafruit_matrixportal_esp32s3.log](https://github.com/user-attachments/files/22076833/adafruit_matrixportal_esp32s3.log)

tannewt, 2025-09-03:
IDF 5.5 removed that function. I've removed it from the latest code here but you may need a release. I don't actually know what version the Arduino IDE uses.

https://github.com/adafruit/Adafruit_Protomatter/blob/f83bac7e421077812523fddb83d3e25f29753315/src/arch/esp32-s3.h#L132

lboue, 2025-09-05:
> IDF 5.5 removed that function. I've removed it from the latest code here but you may need a release. I don't actually know what version the Arduino IDE uses.

I found the answer here: [Arduino Release v3.3.0 based on ESP-IDF v5.5.0](https://github.com/espressif/arduino-esp32/releases/tag/3.3.0)

lboue, 2025-09-05:
> IDF 5.5 removed that function. I've removed it from the latest code here but you may need a release. I don't actually know what version the Arduino IDE uses.

It works fine with the latest code. Could you trigger a new release?

```c++
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
"C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.5\\tools\\gen_esp32part.exe" -q "C:\\Users\\user\\AppData\\Local\\arduino\\sketches\\EE80460EF058C44199F11A701A451B97/partitions.csv" "C:\\Users\\user\\AppData\\Local\\arduino\\sketches\\EE80460EF058C44199F11A701A451B97/pixeldust.ino.partitions.bin"
cmd /c if exist "C:\\Users\\user\\AppData\\Local\\arduino\\sketches\\EE80460EF058C44199F11A701A451B97\\libraries\\Insights" "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.5\\tools\\gen_insights_package.exe" "C:\\Users\\user\\AppData\\Local\\arduino\\sketches\\EE80460EF058C44199F11A701A451B97" pixeldust.ino "C:\\Users\\user\\Documents\\Arduino\\libraries\\Adafruit_Protomatter\\examples\\pixeldust"
cmd /c if exist "C:\\Users\\user\\AppData\\Local\\arduino\\sketches\\EE80460EF058C44199F11A701A451B97\\libraries\\ESP_SR" if exist "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-33fbade6\\esp32s3\\esp_sr\\srmodels.bin" COPY /y "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-33fbade6\\esp32s3\\esp_sr\\srmodels.bin" "C:\\Users\\user\\AppData\\Local\\arduino\\sketches\\EE80460EF058C44199F11A701A451B97\\srmodels.bin"
"C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\4.6/esptool.exe" --chip esp32s3 merge_bin -o "C:\\Users\\user\\AppData\\Local\\arduino\\sketches\\EE80460EF058C44199F11A701A451B97/pixeldust.ino.merged.bin" --fill-flash-size 8MB --flash_mode keep --flash_freq keep --flash_size keep 0x0 "C:\\Users\\user\\AppData\\Local\\arduino\\sketches\\EE80460EF058C44199F11A701A451B97/pixeldust.ino.bootloader.bin" 0x8000 "C:\\Users\\user\\AppData\\Local\\arduino\\sketches\\EE80460EF058C44199F11A701A451B97/pixeldust.ino.partitions.bin" 0xe000 "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.5/tools/partitions/boot_app0.bin" 0x10000 "C:\\Users\\user\\AppData\\Local\\arduino\\sketches\\EE80460EF058C44199F11A701A451B97/pixeldust.ino.bin"
esptool.py v4.6
Wrote 0x800000 bytes to file C:\Users\user\AppData\Local\arduino\sketches\EE80460EF058C44199F11A701A451B97/pixeldust.ino.merged.bin, ready to flash to offset 0x0
Using library Wire at version 3.0.5 in folder: C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\Wire 
Using library Adafruit LIS3DH at version 1.3.0 in folder: C:\Users\user\Documents\Arduino\libraries\Adafruit_LIS3DH 
Using library SPI at version 3.0.5 in folder: C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\SPI 
Using library Adafruit BusIO at version 1.17.2 in folder: C:\Users\user\Documents\Arduino\libraries\Adafruit_BusIO 
Using library Adafruit Unified Sensor at version 1.1.15 in folder: C:\Users\user\Documents\Arduino\libraries\Adafruit_Unified_Sensor 
Using library Adafruit PixelDust at version 1.1.3 in folder: C:\Users\user\Documents\Arduino\libraries\Adafruit_PixelDust 
Using library Adafruit Protomatter at version 1.7.0 in folder: C:\Users\user\Documents\Arduino\libraries\Adafruit_Protomatter 
Using library Adafruit GFX Library at version 1.12.1 in folder: C:\Users\user\Documents\Arduino\libraries\Adafruit_GFX_Library 
"C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp-xs3\\2302/bin/xtensa-esp32s3-elf-size" -A "C:\\Users\\user\\AppData\\Local\\arduino\\sketches\\EE80460EF058C44199F11A701A451B97/pixeldust.ino.elf"
Sketch uses 383489 bytes (18%) of program storage space. Maximum is 2097152 bytes.
Global variables use 31444 bytes (9%) of dynamic memory, leaving 296236 bytes for local variables. Maximum is 327680 bytes.
```

closed 2025-09-05 by tannewt